starnight/LoRa #8: Can't make module on raspberry pi3

YandievRuslan, 2018-03-13:
when i type "make"...
make -C /lib/modules/4.9.59+/build M=/tmp/LoRa/LoRa modules
make[1]: Entering directory '/lib/modules/4.9.59+/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/lib/modules/4.9.59+/build'
Makefile:8: recipe for target 'all' failed
make: *** [all] Error 2

starnight, 2018-03-14:
Which operating system do you install on your Raspberry Pi 3?  Raspbian or some else?
The one I use is Archlinux ARM, not the Raspbian.  But, I think this codes could be build in any most Linux distribution, except the kernel changing problem.  For example, 4.15 is big difference than ever before.

Maybe, you can check the "linux-headers" package is installed or not.  I am not sure it is also called "linux-headers" or some name else in Debian series.  But, I guess you can try to check the kernel module development related packages are installed first.

YandievRuslan, 2018-03-15:
Yes, i use Raspbian.
And yes you are right, i had to install "raspberrypi-kernel-headers" packet.
Now is all right.
Tnks;

closed 2018-03-15 by starnight