starnight/LoRa #11: How can I test the lora driver？

kp339, 2018-06-12:
Hi, I just cross compile the driver and install successful, if I type "ifconfig" cmd I can found the wpan0 site like this
`[root@TW-G01] /mnt/sda1/twiot/ko/lora# ifconfig 
eth0      Link encap:Ethernet  HWaddr EE:58:73:D1:B7:C3  
          inet addr:192.168.1.108  Bcast:192.168.1.255  Mask:255.255.255.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

eth1      Link encap:Ethernet  HWaddr BA:93:D3:5D:F4:F0  
          inet addr:192.168.0.108  Bcast:192.168.0.255  Mask:255.255.255.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

wpan0     Link encap:UNSPEC  HWaddr E0-BB-B3-C3-DF-CA-88-87-00-00-00-00-00-00-00-00  
          UP BROADCAST RUNNING NOARP  MTU:127  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:300 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
`
How can I test the communication between two devices? Can anyone tell me the steps to use the test application? Thanks!

kp339, 2018-06-13:
I will follow the test application's README file and have a try, thanks!

kp339, 2018-06-26:
Hi, I just follow the README file and have a try, but the socket always bind failed, here is the log:
`[root@TW-G01] ~/test/lora# ip addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue 
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast qlen 1000
    link/ether 16:f4:10:0d:01:aa brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.108/24 brd 192.168.1.255 scope global eth0
       valid_lft forever preferred_lft forever
3: eth1: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast qlen 1000
    link/ether 8e:48:bb:ab:27:57 brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.108/24 brd 192.168.0.255 scope global eth1
       valid_lft forever preferred_lft forever
4: sit0@NONE: <NOARP> mtu 1480 qdisc noop 
    link/sit 0.0.0.0 brd 0.0.0.0
5: wpan0: <BROADCAST,NOARP,UP,LOWER_UP> mtu 127 qdisc pfifo_fast qlen 300
    link/[804] 2b:46:df:2b:4c:d8:5d:c5 brd ff:ff:ff:ff:ff:ff:ff:ff
6: lowpan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1280 qdisc noqueue 
    link/[825] 2b:46:df:2b:4c:d8:5d:c5 brd ff:ff:ff:ff:ff:ff:ff:ff
    inet6 fe80::2946:df2b:4cd8:5dc5/64 scope link 
       valid_lft forever preferred_lft forever
[root@TW-G01] ~/test/lora# ./server fe80::2946:df2b:4cd8:5dc5 8080
srv_ip is fe80::2946:df2b:4cd8:5dc5
srv_port is 8080
bind socket failed: Invalid argument
[root@TW-G01] ~/test/lora# 
`
Do anyone have the same problem with me？How can I solve this problem? 
Thanks！

starnight, 2018-06-26:
_fe80::2946:df2b:4cd8:5dc5/64_ is a **Link-local** IPv6 address.  According to IPv6 spec, you have to assign it to wich interface.

Here is the example.  You can refer to page 76 of the slide https://www.slideshare.net/chienhungpan/lets-have-an-ieee-802154-over-lora-linux-device-driver-for-iot