starnight/LoRa #13: crosscompile against 4.19

frank-w, 2019-02-04:
Hi,

can you guide me cross-compiling the module for armhf against Linux 4.19?

i try to integrate it to https://github.com/frank-w/BPI-R2-4.14/tree/4.19-main

i have put it into utils/lora and modified Makefile

```
KERNEL_LOCATION=../..
BUILDDIR=$(KERNEL_LOCATION)
```

and tried running make like this:

`make CROSS_COMPILE=arm-linux-gnueabihf- ARCH=arm `

regards Frank

starnight, 2019-02-08:
This project makes the sx1278 as an external module which does not exist in original Linux kernel.  So, I have not tried the cross-compiling for it.  However, you can build it on an arm board manually, just like on x86 computers.

Or, you can move the source file into Linux kernel source code by yourself for experiment.  Then, also prepare and modify the related Kconfigs and Makefiles which enables to choose the sx1278 module in `make menuconfig`.

frank-w, 2019-02-08:
i compile my kernel with crosscompile (don't want doing on target-machine) so i added it to lora makefile too, as described above. i also changed path to my kernel-source...i know that it is not existing in kernel itself, but makefile needs kernel-sources so i only changed the path to them

problem is that module seems not to be compatible with 4.19, many missing, duplicate and different definitions

have compiled it against 4.19 kernel-sources yet?

starnight, 2019-02-08:
Please show the compiling error.

BTW, have you tried the [commit](https://github.com/starnight/LoRa/commit/0ef184536768ee75992f8c26f7b036b4fc986b11) which is just pushed few hours ago?  I have tried it on my laptop whose kernel is 4.20.

frank-w, 2019-02-08:
have now added the patch and got this result:

```
make -C ../.. M=/media/data_ext/bpi-r2-kernel/github/utils/lora modules
make[1]: Entering directory '/media/data_ext/bpi-r2-kernel/github'
  CC [M]  /media/data_ext/bpi-r2-kernel/github/utils/lora/sx1278.o
  Building modules, stage 2.
  MODPOST 1 modules
WARNING: "ieee802154_xmit_complete" [/media/data_ext/bpi-r2-kernel/github/utils/lora/sx1278.ko] undefined!
WARNING: "ieee802154_rx_irqsafe" [/media/data_ext/bpi-r2-kernel/github/utils/lora/sx1278.ko] undefined!
WARNING: "ieee802154_register_hw" [/media/data_ext/bpi-r2-kernel/github/utils/lora/sx1278.ko] undefined!
WARNING: "ieee802154_alloc_hw" [/media/data_ext/bpi-r2-kernel/github/utils/lora/sx1278.ko] undefined!
WARNING: "ieee802154_free_hw" [/media/data_ext/bpi-r2-kernel/github/utils/lora/sx1278.ko] undefined!
WARNING: "ieee802154_unregister_hw" [/media/data_ext/bpi-r2-kernel/github/utils/lora/sx1278.ko] undefined!
  CC      /media/data_ext/bpi-r2-kernel/github/utils/lora/sx1278.mod.o
  LD [M]  /media/data_ext/bpi-r2-kernel/github/utils/lora/sx1278.ko
make[1]: Leaving directory '/media/data_ext/bpi-r2-kernel/github'
```

starnight, 2019-02-08:
The log says sx1278 module is built now.  Please copy it to `$(KERNEL_LOCATION)/build/` of your target board.

All of the functions in the warnings are provided by the **mac802154** module in Linux kernel.  You can check the **mac802154** is built within the Linux kernel or not for the warnings.  If it is not, please toggle it to Y or M in `make menuconfig` when you build Linux kernel.

frank-w, 2019-02-08:
thank you, this fixes that

now i'm hanging on overlay (have changed target-dir and removed sudo)...

```
dtc -I dts -O dtb -@ -o rpi-lora-spi.dtbo rpi-lora-spi-overlay.dts
rpi-lora-spi.dtbo: Warning (unit_address_vs_reg): Node /fragment@0 has a unit name, but no reg property
rpi-lora-spi.dtbo: Warning (unit_address_vs_reg): Node /fragment@0/__overlay__/spidev@0 has a unit name, but no reg property
rpi-lora-spi.dtbo: Warning (unit_address_vs_reg): Node /fragment@0/__overlay__/spidev@1 has a unit name, but no reg property
```

starnight, 2019-02-09:
The original issue is solved.
The overlay question is another issue.  Please separate it to a new ticket.

frank-w, 2019-02-09:
Ok,have opened a new issue for dts

Where should i put the module? build-symlink is invalid on target

starnight, 2019-02-09:
Please copy the built module to `/lib/modules/<kernel-version>/extra/` of your target board.
It is described in [Building External Modules](https://www.kernel.org/doc/Documentation/kbuild/modules.txt)

closed 2019-02-09 by frank-w