starnight/LoRa #18: Is it compatible with Dragino LoRa / GPS HAT?

tmxk4221, 2019-09-26:
I'm using Dragino LoRa / GPS HAT which includes sx1276/8.

I installed and loaded the device driver according to the manual posted by the author.

But the LoRa module does not work properly.

Are there any settings other than the manual posted by the author?

starnight, 2019-09-28:
This experimental project is IEEE 802.15.4 MAC over LoRa phy, not LoRa MAC over LoRa phy.  So, it cannot communicate with original LoRa gateway.

tmxk4221, 2019-09-29:
Thank you for your answer !! But I did not understand well.
I'm not sure what the original LoRa gateway is.
I used two Dragino LoRa / GPs HATs with sx1278 and two Raspberry pi 3B loaded with rasbian 4.14.
I have installed device drivers on two raspberry pis and want to communicate between them.