starnight/LoRa #19: Not detecting RF output

DanielRIOT, 2019-11-13:
I've got a Hoperf RF95 ( SX1276 compatible module ) connected to a Beaglebone SPI port , Driver loads, wpan0 and lowpan0 is created.
dmesg output shows

>  sx1278 spi2.0: add an IEEE 802.15.4 over LoRa SX1278 compatible device

 but when client is run ( or ping6 ) I do not see ant RF on the spectrum analyser.

iwpan phy phy0 set tx_power also reports as 
"command failed: Not supported (-95)"

Devicetree 

> sx1278@0{
> compatible = "semtech,sx1276";
> spi-max-frequency = <2000000>; 
> reg = <0>;
> clock-frequency = <32000000>;
> center-carrier-frq = <868000000>;
> rf-bandwidth = <125000>;
> spreading-factor = <7>;
> minimal-RF-channel = /bits/ 8 <0>; // 0 :  868.3 MHz
> maximum-RF-channel = /bits/ 8 <10>; // 1-10 :902 - 928 MHz
> };

iwpan phy0 info output :

> wpan_phy phy0
> supported channels:
>         page 0: 0,1,2,3,4,5,6,7,8,9,10
> current_page: 0
> current_channel: 0, 868.3 MHz
> capabilities:
>  iftypes: node,monitor
> channels:
>                 page 0:
>                         [ 0] 868.3 MHz, [ 1]   906 MHz, [ 2]   908 MHz,
>                         [ 3]   910 MHz, [ 4]   912 MHz, [ 5]   914 MHz,
>                         [ 6]   916 MHz, [ 7]   918 MHz, [ 8]   920 MHz,
>                         [ 9]   922 MHz, [10]   924 MHz
> min_be: 3
> max_be: 5
> csma_backoffs: 4
>  frame_retries: 3
>  lbt: false

iwpan dev wpan0 info

> Interface wpan0
>         ifindex 3
>         wpan_dev 0x1
>         extended_addr 0x8adb36432912159c
>         short_addr 0xffff
>         pan_id 0xbeef
>         type node
>         max_frame_retries 3
>         min_be 3
>         max_be 5
>         max_csma_backoffs 4
>         lbt 0

When I run client and expect to see RF emissions but none are seen on spectrum analyser I do not see any 

> client fe80::88db:3643:2912:159c%lowpan0  fe80::88db:3643:2912:1
> 69d 200 test
> Send test with in 4 bytes

Do you know why this could be ?

starnight, 2019-11-14:
I remember Hoperf has different register addresses and meanings.  So, the driver is not compatible.

DanielRIOT, 2019-11-15:
I managed top get  HopeRF RFM95W to work ( the new datasheet  -V2- has a correct register map - which matches the sx127x )
, My module is transmitting slightly low ( expecting 868.3, but getting 867.980 ) but that is probably a calibration or LO setting issue that I can solve..
[image]

Do you know why " iwpan phy phy0 set tx_power " would report as "Not supported (-95)"  no matter which power value one tries to set ?

-Daniel

closed 2019-12-05 by DanielRIOT